sinara-hw/DiPho #1: [RFC] Digital monitoring photodiodes

airwoodix, 2021-02-10:
Continuing the discussion in sinara-hw/meta#52 ([comment](https://github.com/sinara-hw/meta/issues/52#issuecomment-624352053)) on cheap monitoring photodiodes for Sinara.

  - Something like Thorlabs [S120C](https://www.thorlabs.com/thorproduct.cfm?partnumber=S120C)
  - **form factor**: max 1"-ish diameter, 20 mm deep (see Thorlabs link above)
  - **photodetector**: Si photodiode, area > 2x2 mm², e.g. [BPW34](https://www.vishay.com/docs/81521/bpw34.pdf)
  - **bandwidth**: tens of Hz to few kHz, more not relevant
  - **photocurrent**: 10 µA to 10 mA
  - **gain**: if possible linear, selectable in decades, if not, log amplifier
  - **interface**: digital, SPI-like, ideally single cable for power and data to logging board in Sinara crate. RJ45 is too bulky, USB-C sounds cool but doesn't lock
  - **cable length**: around 2 meters would be convenient

Does this cover enough use cases?

I only see opportunities if this can be around 10 times cheaper than the Thorlabs product, so max 50 €/$ per channel.

gkasprow, 2021-02-10:
@jordens do you have the mechanical files for your SM1 design?

jordens, 2021-02-10:
[logpd.gerber.zip](https://github.com/sinara-hw/DiPho/files/5960682/logpd.gerber.zip)
[logpd.view.pdf](https://github.com/sinara-hw/DiPho/files/5960683/logpd.view.pdf)
[logpd.sch.pdf](https://github.com/sinara-hw/DiPho/files/5960684/logpd.sch.pdf)

pm-daniel, 2021-02-15:
Hi, I'm a student who will be doing that project. I'd prefer sticking to the STM32 platform as an MCU cause I've far more experience working on them rather than ESP32 or other platforms. If there is a necessity for that sensor to have wireless communication of some flavor then I would opt for STM32 + wifi/bt module. As far as wired communication goes, I understand that there should be an M8 connector (ethernet, MQTT protocol) and USB-C for console (configuration/setup?; straight forward TX/RX pair or USB VCP?). What about RS-485? Are we ditching that idea? In terms of mechanical design, I understand that project should fit on two circular PCB's (~24mm dia) - am I right ? Are there any other things that I should take into consideration?

gkasprow, 2021-02-19:
Once we have the Ethernet, RS485 makes not much sense.
Yes, we can use USB VCP. The design should be cheap, so before we freeze some decision, the budget must be made.
The design should fit in the enclosure. Since we want analog output, it would be probably easier to use two rectangular boards.

gkasprow, 2021-02-19:
We could use a low-cost CPU like STM32F072CBU6 which has a USB and the package is tiny.

dhslichter, 2021-02-19:
not sure why analog output means that rectangular boards are better?  The point of circular boards is to make it possible to fit them inside a 1" lens tube for the photodiode application -- this makes the mounting mechanics simple and easy.

gkasprow, 2021-02-19:
You have to somehow expose the coax connector. It can of course protrude through the rear digital board, but it occupies some space. If we place the board along the sensor axis we have access to rear connectors but have issues with sensor fixing (must use THT) and smaller board area.
Anyway, after the selection of enclosure and connectors and initial placement, we need to make some mockups to see what fits and what doesn't.

gkasprow, 2021-02-19:
I did a quick mockup. It looks like we can easily fit the entire functionality on a single PCB. M8 connectors are far too big. But we can use a USB-C connector. It has normally unused high-speed lanes that can be adapted for Ethernet.
In such a way if we connect it to a standard USB hub, the circuit will enumerate as a serial port of whatever we implement and will be supplied over USB
If we plug it into a dummy 5V power injector, using either USBC-RJ45 cable or a dedicated passive adapter, it will work as an MQTT client. We can even skip the LAN transformer because we don't need isolation. It will always be one on the switch side.
Here are some mockups
version with LAN trafo
[image]

Thormalbs S120C-like enclosure
[image]

ADC, Logamp, diode, STM32, ENC424J600 PHY
[image]

If we want to add analog output, there are problems with such high SMA connectors - I found only one (AMP 132291) that fits somehow.
If we skip the LAN trafo, we can fit the SMA connector. There are also super-small LAN [trafos avaialble](https://product.tdk.com/info/en/catalog/datasheets/090007/trans_alt_en.pdf)
[image]
[image]
[image]

and the possible final effect
[image]
[image]
[image]
[image]

gkasprow, 2021-02-19:
Of course, we can make a dedicated sensor board, but the supply and SMA would have to go from the rear board. SMA can also protrude through the digital board, but the analog board would need another connector with a cable for the power supply. It's easier to leave the SMA and USB on the digital board and if someone wants a cheap solution, simply don't mount the CPU, LAN PHY, and ADC and treat the digital board as power delivery.
To connect the two boards one can use 1.27mm gold pin connectors or i.e. Molex SlimStack™ B2B connectors.

gkasprow, 2021-02-19:
@pm-daniel 
the PHY circuit is here https://github.com/sinara-hw/Booster/tree/master/PCB/Booster_User_Interface_ENC424J600
the CPU is here https://github.com/elhep/SiPM_AFE
I published initial files to the repo. This will give you a kick-start.

gkasprow, 2021-02-20:
The wireless connectivity can be easily provided using RPI0 and USB cable.

gkasprow, 2021-02-20:
two-board version
[image]
[image]

gkasprow, 2021-02-20:
[image]

gkasprow, 2021-02-20:
Do we need more than 12Bit ADC? If not, then one from STM32 can be used. We also have two 12b-bit DACs that can be used for offset correction.

pm-daniel, 2021-02-20:
the two-board design seems more versatile and should be easier to design. If we decide to go with two boards then we should be able to fit M8 connector which footprint is actually smaller than the footprint of SMA connector (12.5x10mm vs 13x13mm). In my mind another benefit of the two-board design is better separation between analog and digital sections, am I right?

jordens, 2021-02-20:
12 Bit is plenty.

gkasprow, 2021-02-20:
M8 is much bigger than SMA. here is the comparison 
[image]

gkasprow, 2021-02-20:
USB is free, for M8 you have to pay.

gkasprow, 2021-02-20:
We have 50$ budget including mechanics and assembly. M8 is 15$ or so.

gkasprow, 2021-02-20:
And we can make a simple USBC to RJ45 adapter placed in a more convenient place if somebody doesn't want to make a cable. Such an adapter would fan out the USBC to USBC for supply/USB and RJ45 for Ethernet. That would be super-cheap.

gkasprow, 2021-02-20:
we can try to use TL061 opamp - they are super-cheap. The offset can be canceled using MCU DAC. We don't need external ADC so we can make it very simple from an HW point of view. Gain can be switched using a low-cost 4051mux. All gain corrections would be made in SW. You can also find a lower-cost B2B connector.

gkasprow, 2021-02-20:
USB-C cable can be built in a few minutes using a piece of RJ45 patchcord and the plug
[image]

gkasprow, 2021-02-20:
the KSZ8851 MAC+PHY over SPI has Auto-MDIX so we don't care about Ethernal Rx/Tx pair swapping

gkasprow, 2021-02-20:
Since we have a dedicated AFE board, let's place both TIA and log amp and use the assembly option to connect one or another to the photodiode. We have enough ADC inputs to connect both outputs.

gkasprow, 2021-02-20:
KSZ8851 is also much smaller than the ENC chip, so we have more place on the digital board. This particular MCU has also a CAN controller but I'm not sure if it's worth it to place a 0.5$ transceiver and use the remaining USB-C pins for such connectivity.

pm-daniel, 2021-02-20:
fair point about M8 being larger than SMA - I forgot about that bulky mechanical part of the connector and made that statement based on the inner part drawing only:
[image]
[image]

gkasprow, 2021-02-20:
[Here](https://www.analog.com/en/analog-dialogue/articles/programmable-gain-transimpedance-amplifiers.html) is an article on making the programmable gain TIA.

gkasprow, 2021-02-20:
And some hints on TIA optimization:
https://www.analog.com/en/technical-articles/optimizing-precision-photodiode-sensor-circuit-design.html

pm-daniel, 2021-02-20:
thanks, looks like I have some "homework" to do

pm-daniel, 2021-02-28:
I've created the first version of a simple block diagram for DiPho
[DiPho_block_diagram.pdf](https://github.com/sinara-hw/DiPho/files/6057235/DiPho_block_diagram.pdf)

gkasprow, 2021-03-14:
@pm-daniel 
- STM32G4 is nearly 2x more expensive and overkill for this application. For differential ADC, you need a differential driver. It may be not worth the circuit complication. The TIA will be spaced by 1..2cm from ADC, I don't see a clear advantage of differential signalling here.
- I'd avoid any DC/DC converters here, especially PoE flyback converters as they are bulky and it's difficult to filter EMI they crate.
- PoE can be entirely passive. Active one can be without isolation but would create ground loop issues. Isolated one is bulky.
- I would use transformer-less Ethernet and make a power supply injector as a tiny PCB with USB-C (DiPho), USB-C(power brick) and RJ45. One can make one with USB-C, RJ45 and proper isolated PoE  and use when needed. In this way we have 3 usage scenarios - first with cheap USB-C to RJ45 cable. Supply can be delivered using cheap POE injector or directly from the power suply. More elegant solution is an adapter where you plug the USB charger and RJ45 cable. Third one is 802.11af compliant power solution.  The advantage is that the basic configuration is simple and cheap but requires custom cable which can be easily made by soldering USB-C and RJ45 cables.

gkasprow, 2021-03-14:
@pm-daniel when can I expect the first version of schematics?

pm-daniel, 2021-03-15:
@gkasprow 
- ok, then I will use STM32F072C8U and skip adc diff driver step.
- I like the approach with custom cable with tiny PCB - it basically means that we can provide power to DiPho via USB power lines, so what's the point in using PoE? Maybe I misunderstood something?
[image]
- I will try to finish a schematic draft by the end of march

pm-daniel, 2021-03-15:
But are there any standardized voltage/current values for passive PoE? I found some info about 24V/48V but it seems pretty high with no DC/DC converter onboard...

gkasprow, 2021-03-15:
Passive PoE is always custom. There is no standard behind it. People are using usually 12V but there are also 5V versions. You simply use free RJ45 lines or transformer taps to transfer DC. Some are using the bridge rectifiers to make it slightly more robust.
In our case we have a way to supply 5V via USB, so let's add a proper PoE converter to the adapter PCB. If someone needs PoE, will simply install such a converter. We are using Silvertel converters, but there are also lower-cost ones used in Arduino clones.

gkasprow, 2021-03-15:
The only difference between standard PoE and passive PoE is the lack of a power negotiation mechanism. Voltage is a secondary thing, standard PoE modules can sometimes work with voltages down to 12V, but it is usually 48V.

pm-daniel, 2021-03-15:
Ok. Thanks for that clarification.
By proper PoE converter on that adapter PCB you mean Active PoE compatible with 802.3af/at or just DC/DC converter with max Vin >=48V? 
And as I understand, power would run from the adapter PCB to DiPho via VCC/GND pins on USB-C connectors, because PoE conversion will happen on the adapter PCB. But then eth transformer should also be located on that adapter PCB, am I right? 
So in the end adapter PCB, will have similar functionality (+external power input) to this PoE splitter, but RJ45 and USB cables will be merged into one cable? 
[image]

gkasprow, 2021-03-15:
Yes, I mean using for exampel Silvertel converter + RJ45 connector. Look how it is done in Thermostat design.

pm-daniel, 2021-03-31:
@gkasprow I've just pushed the first revision/draft schematics of afe and mcu boards.

gkasprow, 2021-04-30:
@pm-daniel please use the master branch. My files were just templates.
Did you do some simulations of the AFE?
Why this particular opamp was used?

gkasprow, 2021-04-30:
It's a bad idea to use both SSTX pairs for Ethernet signals - you will get stubs. That's why we use Auto MIDIX to enable automatic pair detection. Just use SSTX 1 and SSTX2.

gkasprow, 2021-04-30:
try to use a standard SWD connector, of course, if that fits

gkasprow, 2021-04-30:
please use ports only for multi-page projects. for local signals use nets. See how other Sinara projects are organized

pm-daniel, 2021-05-07:
> @pm-daniel please use the master branch. My files were just templates.
Of course - I pushed to the default git branch (which now happens to be "main" instead of master), I will rearrange it and push the next commits to "master"
> Did you do some simulations of the AFE?
Not yet, If I manage I will do some simulations this weekend. I understand that I should use the current source as a stimulus?
> Why this particular opamp was used?
I choose OPA141 because it has significantly smaller input noise and input bias current, is available in a smaller package than TL071 f.e. and its price is not horrible IMO.

pm-daniel, 2021-05-07:
> try to use a standard SWD connector, of course, if that fits

I'll try to fit it

gkasprow, 2021-05-09:
Just make sure the file you push is listed on Github.

 >I understand that I should use the current source as a stimulus?

yes, but please add photodiode mode.

>I choose OPA141 because it has significantly smaller input noise and input bias current, is available in a smaller package than TL071 f.e. and its price is not horrible IMO.

right. But in our case, you have to prove you need low voltage input noise. We are working with high-value feedback resistor so the current noise may be dominating.  Low input bias current is not that critical since you can easily compensate for it. But be careful because you can add some noise! Here is an [online tool ](https://www.jensign.com//transimpedanceamp/index.html)to play with TIA. Essentially, you have to show that you know what you are doing :)

pm-daniel, 2021-05-14:
@gkasprow sure. 
It's a busy time right now but I'll try to do some simulations (and documents) in the following week or two

pm-daniel, 2021-06-06:
@gkasprow After doing a few simulations it became obvious that OPA141 may not be the best pick (TL071 either). Because of the pretty bad output voltage swing characteristic - I totally forget about it (for TL071 it's +-1.5V from the supply voltage and +-0.3V from the supply voltage for OPA14). I've tried a few different opamps (AD8541, AD8539, ADA4891) and ADA4891 looks most promising and it has a very low input bias current, very good swing characteristic, and a similar price to OPA141. All those conclusions are based on transient simulations that I conducted in LTspice.
I used the current source with parallel capacitance as stimulus and now I'm struggling with noise simulation as it looks like LTspice expects voltage source to be selected as a noise source. Is there a better way of simulating photodiode input in the LTspice? I've spent some time on research and it looks like everyone uses just the current source with capacitance (just like I did). 
Once I complete the noise and bandwidth simulation I will publish results on GitHub. Maybe you have any other recommendations regarding opamp choice?

dnadlinger, 2021-06-06:
A command like `.noise V(OUT) I1 dec 1000 10 10Meg` (where OUT is the output node, and I1 is the current source) should work fine in LTspice.

pm-daniel, 2021-06-06:
@dnadlinger thank you! After seeing your reply I've checked my simulation command again and it turns out that I had IN and OUT nodes defined in the wrong order. It works just fine now.

gkasprow, 2021-08-17:
@pm-daniel can you open a new issue and post your noise simulations?

airwoodix, 2022-07-14:
@pm-daniel Do you have an update on this project? How did the testing of the first boards go? Thanks!

pm-daniel, 2022-07-14:
@airwoodix Tests are not finished yet - they will be ready by the end of the month. I'll publish the results on GitHub. Sorry for the delays but in the meantime, I'm writing my BEng thesis. Working a full-time job at Creotech (on non-related projects) also doesn't help. I hope you understand!

airwoodix, 2022-07-14:
Of course! :slightly_smiling_face: Thanks a lot for the update. I'm looking forward to the testing results.

gkasprow, 2022-07-26:
@pm-daniel can you post some photos of the PCBs?

pm-daniel, 2022-07-29:
[image]
[image]

It's mounted in measurement fixture for now so those are the best photos I can take at the moment

I'll upload better ones if needed after finishing the measurements