sinara-hw/DiPho #2: [RFC] Other front end boards

dtcallcock, 2021-02-18:
Right now it looks like DiPho will be a two board device, with one board having the photodiode + amp, and one board having the DAC, uC, power, and communications.  It has already been mentioned that different versions of the photodiode board could be produced in the future (eg. with log amp). One could imagine a much broader range of applications though (eg. analog front end for voltage monitoring/logging, environmental monitors).  I mention it now as, if this is thought to be a good idea, it may influence exactly how things are divided between the two boards and what kind of header should go between them.

dtcallcock, 2021-02-18:
Rf power meter + frequency counter is another idea.

gkasprow, 2021-02-18:
true, it could be a laser driver, humidity sensor,  single channel Sampler, optical power meter, etc. I would expose all pins we have

dhslichter, 2021-02-19:
is this too much mission creep?  there are a million little commodity IoT widgets out there.  I don't think it hurts to expose all available pins, but maybe better to focus on making this a good photodiode monitor first?

dtcallcock, 2021-02-19:
> better to focus on making this a good photodiode monitor first?

I'd argue that the photodiode aspect of this project is trivial.  The 'IoT widget' aspect seems to be the tricky part and thus I think we should consider leveraging effort put in there.

> is this too much mission creep?

I think this issue is more about keeping this in mind and not doing anything to preclude it unless there is good reason, rather than creeping the mission.

dhslichter, 2021-02-19:
> I'd argue that the photodiode aspect of this project is trivial. 

Agreed.

> I think this issue is more about keeping this in mind and not doing anything to preclude it unless there is good reason, rather than creeping the mission.

Fair.  Just wanted to raise the point.  It's already starting to look like the all-singing, all-dancing IoT widget with 4 different communication protocols and 3 connectors and power harvesting and....