sinara-hw/DiPho #3: [RFC] TIA Simulations

pm-daniel, 2021-08-22:
@gkasprow finally I've managed to finish TIA simulations (attached pdf file)  for several different opamp types. As that's my first time doing noise (and TIA in general) simulations in LTSpice there could be some errors that I'm not aware of.

[DiPho_TIA_Simulations.pdf](https://github.com/sinara-hw/DiPho/files/7027212/DiPho_TIA_Simulations.pdf)

gkasprow, 2021-08-22:
There is something wrong with TL071 and OPA141. Did you apply proper voltage? Can you post schematic?
When designing TIA, one should balance current noise and voltage noise contribution to the noise budget.
Did you do the noise budget?

pm-daniel, 2021-08-22:
Regarding voltage, I applied a 5V single-rail and it turns out to be sufficient for all tested opamps but TL071 which needs min 7V (but we have max 5V on the AFE board). Then it's no surprise that the results for TL071 are wrong. In the case of OPA141, I checked once again and the voltage should be sufficient, and I tested all opamps in the same schematic.  In my simulation, I've selected the current source as input, so I reckon that voltage noise is not taken into account, am I right? In that case what additional simulations I should make?

Here is the schematic I used for simulations:
[image]

closed 2022-12-05 by pm-daniel