sinara-hw/DiPho #10: PCB review

gkasprow, 2021-11-20:
Digital board
- doesn't this particular STM MCU require crystal for USB operation?
- Use L1, D1 the same as in other Sinara project (Stabilizer, Thermostat)
- please add 10k pullup to NRST. I'm not sure C4 is good idea, it may interfere with debugger. If such one is on the devkit than it's fine.
- Is C31 sufficient for this LDO?
- use 1mm/0.15mm silkscreen, enable only active components. there is usually no place for all RCs
- missing board shape on Mechanical1
- wouldn't it be better to use SPI bus between MCU and Ethernet chip? TCP/IP stacks usually use such interface because it works with DMA.  This STM can't expose internal parallel data (EMIF) bus via IO pins. This means that each IO transfer would have to be done in SW which will add a lot of overhead. Just look how stacks handle external Ethernet PHYs
- can VBAT be leaft floating?
- we need an unique MAC address. Use the same MAC EEPROM chip as in Stabilizer/Thermostat

Analog board
- use 1mm/0.15mm silkscreen, enable only active components. there is usually no place for all RCs
- missing board shape on Mechanical1

Adapter
- move USB to the board edge
- adapt PCB shape to some existing and available enclosure, preferably [one with panels.](https://www.kradex.com.pl/product/obudowy_specjalne_inne_z_miejscem_pod_wyswietlacz/z67?lang=pl) Such panels can be easily cut with laser ploter.
- add a PCB placeholder for EMI filter (fig 19 in datasheet)
- use RA LED close to panel, preferably same as in other designs
- why two LEDs are used?
- add another supply connector, preferably USB so it would be possible to use Ethernet without PoE
- instead of D1 use the same diodes as in Thermostat.
- use 1mm/0.15mm silkscreen, enable only active components. there is usually no place for all RCs
- missing board shape on Mechanical1

gkasprow, 2021-11-20:
@pm-daniel I finished the review.

gkasprow, 2021-11-20:
- [ ] avoid via in the pad, add Altium rule. 
[image]
- [ ] some components are too close. Try to use Mechanical 15 rectangles as a clearance marker 
[image]
- [ ] tent vias, but only ones that are not exposed on the thermal pads.

pm-daniel, 2021-11-27:
@gkasprow I fixed all mentioned things in the adapter project 077a311

pm-daniel, 2021-11-29:
@gkasprow fixed a few things in the digital board schematic, have a look a5d5d1f

gkasprow, 2021-12-11:
@pm-daniel go on with the PCB design

pm-daniel, 2021-12-16:
@gkasprow I've finished AFE PCB f7f5641

gkasprow, 2021-12-17:
ok, looks good, but there are still a few issues
- [ ] Avoid such traces. They confuse the AOI
[image]
[image]

Use these instead
[image]
[image]

- [ ] Add and use acute angle rule
- [ ] there are still many DRC issues; add thermal pads to be allowed for vias 
- [ ] digital PCB has many missing connections.

gkasprow, 2021-12-17:
and please mark all the issues above when fixed

pm-daniel, 2021-12-18:
@gkasprow both boards are finished 3fac3f8 26e2a46

gkasprow, 2021-12-18:
oh, I looked at the wrong releases.

pm-daniel, 2021-12-18:
not really, I've pushed them today ;)

pm-daniel, 2022-01-03:
@gkasprow Is there anything else to be fixed?

gkasprow, 2022-01-09:
It looks fine. we can produce prototypes.

pm-daniel, 2022-01-09:
Do I have to prepare production files?

gkasprow, 2022-01-09:
Yes, send them to Technoservice (TSK), and send me an invoice.

airwoodix, 2022-01-17:
@gkasprow @pm-daniel thanks a lot for the great work on this project! Is there already a timeline available for prototypes and firmware development?

pm-daniel, 2022-01-17:
@airwoodix the prototype was sent into production last week

airwoodix, 2022-01-17:
Awesome, thanks for the quick update! Could you please also upload the schematics' PDF as release? or is it too early for that?

gkasprow, 2022-01-17:
@pm-daniel please make a release as in other projects.

pm-daniel, 2022-01-18:
@airwoodix is there any way in which I can contact you? I would be grateful for some information about use cases for iPhone, it would help write my thesis introduction.

airwoodix, 2022-01-21:
Sure. I believe that my email is available from some (recent) commits in m-labs/artiq.

I don't have a specific application in mind at this stage. The motivation for #1 was to see if one can get small footprint general purpose photodiodes to drop in a laser system and record for a more or less extended period of time ("I see some weird periodic signal on my atoms/ions, can I correlate this to laser power?", "I need to couple a fiber that extends to the neighboring building" (current solutions include a return signal cable or a webcam / your-favorite-messenger video call), "I'd like an automatic alert when this laser diode dies, this fiber/cavity coupling gets bad, etc."). One can surely do fancier things depending on the actual bandwidth and dynamic range. The analog output could be used for feedback, as [mentioned](https://github.com/sinara-hw/DiPho/issues/1#issuecomment-772990559) by @dhslichter for example.

IIRC there's a bit more in #1. Overall, I find @jordens' [vision](https://github.com/sinara-hw/DiPho/issues/1#issuecomment-625387512) really attractive and DiPho looks like a nice step in this direction.

airwoodix, 2022-01-28:
@pm-daniel do you have an update on the PDF release? Thanks!

pm-daniel, 2022-01-28:
@airwoodix It will be ready by Monday - I had a really busy week.

pm-daniel, 2022-01-30:
@airwoodix pdfs are ready, there should be pre-release tag visible in repository

airwoodix, 2022-01-31:
@pm-daniel Wonderful, thanks a lot!

closed 2022-03-06 by pm-daniel