sinara-hw/DiPho #19: DiPho v2 ready for review

pm-daniel, 2022-12-01:
@gkasprow I've just pushed the final changes to second revision of the project. It's ready for review and hopefully prototype manufacturing.
Additionally I designed a breakout board which would be helpful during potential hardware debugging and firmware development phases. Picture below.
[image]

gkasprow, 2022-12-20:
- [ ] use holes in the debug board - it's easier to attach probes without damaging SMT pads
- [ ] what's the point of using IC2B?
- [ ] was the analogue switch leakage current issue resolved?
- [ ] use naming convention as in other SInara projects: PA4V5 -> P4V5A; 
- [ ] there is LOG assembly variant, but the log amplifier is not present on schematics
- [ ] there was a shield used (4 clips). Does the shield exist as a part number, or it's custom-made?

gkasprow, 2022-12-20:
digital board
- [ ] why was OSC1 used instead of crystal? Oscillators are usually more power-hungry and more expensive. can we use CLKO of IC5 to drive the MCU?
- [ ] such configuration causes ERC error. Use ports only to connect sheet hierarchy.
- [ ] why was OSC2 used instead of crystal? Oscillators are usually more power-hungry and more expensive. 
- [ ] I'd pull RSTn low to keep IC5 in reset by default. What's the point of R9?
- [ ] keep power rail naming convention (PFILT3V3)
- [ ] does R10/R11 need to be 0805?
- [ ] I wouldn't say I like how L4 is connected. Are you sure the windings are connected correctly? IMHO there should be 2 individual coils; otherwise, the Ethernet signal will be shorted.

gkasprow, 2022-12-20:
- [ ] remove the PoE adapter project
- [ ] Instead of a PoE adapter I'd make a simple router based on RPI nano, the same NCN chip and Ethernet to USB converter and place it into the same case as before. This adapter would provide power to all devices as well as SPE Ethernet connectivity.

pm-daniel, 2022-12-23:
I used oscillators instead of crystals because constrained space and routing possibilities. In case of oscillators I don't have to route two tracks with caution but only single one (additionally without having to worry so much about track lengths and so on), I chose oscs that are fairly inexpensive.  We cannot use CLKO as it provides fixed 25MHz output and with PLL embedded in STM32 one can not obtain 48MHz clock needed for USB.

RSTn pin of IC5 - external pull-up resistor was used in accordance to IC data sheet, internal pull-up is fairly weak so manufacturer recommends using an external in "noise sensitive applications" (whatever that means in that case). Additionally such a resistor is also used in development board.

L4 is connected as a differential-mode choke, I've seen such a connection in many papers discussing PoDL in SPE. ex. 
[image]
page 8 in [this keynote](https://www.ieee802.org/802_tutorials/2015-11/PoDL_tutorial_1115.pdf)
or in this diagram from HARTING
[image]
Albeit I'm not entirely sure whether such a connection is correct.

pm-daniel, 2022-12-23:
Regarding AFE board - MAX4639 has extremely small leakage current and package that is easy to route while being available quite fairly. 

IC2B was used to "isolate" (not the best wording probably ;) )sensitive offset voltage coming into TIA from ADC an digital stuff (whole TIA offset net is routed inside of shield thanks to that). Also that part was already available so... 

those pads on debug board are meant for keystone 5015 SMD test points but I will change them to THT version

gkasprow, 2022-12-30:
ACK with transformer

gkasprow, 2022-12-30:
go on with routing

pm-daniel, 2022-12-30:
@gkasprow Both boards are routed already ;)

regarding "router" - did you mean raspberry pi compute module? Because I haven't found any info on "RPI nano".
do we want only single SPE port? or ex. two of them (thus single NCN chip or more?)
do we want PoE power or external via barrel jack or usb-c or any other connector? maybe both?
do we want any other connections? ex. shell via uart or sth like that?

gkasprow, 2022-12-30:
I meant "Raspberry Pi Pico W". But there is an issue with USB connectivity.
Essentially you would have to use right angled PCB mounted USB plug to get USB connectivity needed for network adapter. Sth like [here](https://botland.com.pl/rozszerzenia-gpio-i-nakladki-hat-do-raspberry-pi-zero/21702-adapter-raspberry-pi-zero-do-3b-waveshare-22382.html?cd=18298825651&ad=&kd=&gclid=Cj0KCQiAtbqdBhDvARIsAGYnXBPZexVO3te5Q8ayEVJJ48f8cYb-c2sJmjzmtpNEJdqOst9LFPCvp6caAjN3EALw_wcB)
Let's add two SPE ports
USB-C is good enough; when necessary one can use Ethernet to USB-C PoE injector
You can use [this project](https://github.com/michalgaska/home-assistant-controller) as a template

gkasprow, 2022-12-30:
This template project uses PoE so we can copy it.

gkasprow, 2022-12-30:
one can use such connector as well https://thepihut.com/products/solderless-zero-dongle-for-raspberry-pi-zero