sinara-hw/DiPho #21: DiPho casing

szymons90, 2023-05-07:
[DiPho_20230507.zip](https://github.com/sinara-hw/DiPho/files/11413999/DiPho_20230507.zip)

Hi,
I prepared two versions of DiPho casing, both consist of two parts - the base and the main housing. There is a sligth difference in an approach to manufacturing.
In V1, the base can be made only by milling from one surface, while the housing by turning and milling. Both parts are screwed together with 3 x M3 screws. This is a slightly more expensive solution, but lighter - the mechanics weighs about 14.5 g
In V2, the base can also be made only by milling, but the casing only by turning (except for the holes fixing the whole thing, but you don't need a special machine for that, you can do it by hand). In this version I used 4 x M2 screws to connect the whole thing. This solution should be a bit cheaper, but it's also a little bit heavier (18.7 g).
In both versions, the SM1 thread for the Thorlabs tube must be made by a die - you can buy one on aliexpress or ebay. On the other hand, if we had a skilled technologist, maybe after a few tries we could make such a thread on a lathe.
PCBs are attached directly to the base. The problem is that due to the small mounting screws - M1.4 - there are no such standard spacer. I suggest to buy some brass tube fi 2 x 1.6 and cut them into pieces. It will be a bit more annoying to mount, but right now, I don't see any cheap and easy solution here.
@gkasprow , @pm-daniel  I am waiting for your comments. Let me know if you prefer one of these versions or if you want a different approach. Step files are in attachment,

[image]
[image]
[image]
[image]

gkasprow, 2023-05-07:
I like the second idea.
What if we 3D print the base and attach only the threaded tube?

gkasprow, 2023-05-07:
I think we need the photodiode surface much closer to the edge, as in the original Thorlabs module.

gkasprow, 2023-05-07:
For mounting hole we could use threaded insert

szymons90, 2023-05-07:
1. Sure, if there are not any special environmental requirements, I don't see any problems.
2. Ok, I'll fix it in next revision
3. Ok, I'll modify the design, in which mounting threaded holes will be in the 3D printed base. The housing will have just an SM1 thread to the tube.

gkasprow, 2023-05-07:
are you sure 3D print technique can print such thread? Would it be durable? I'd opt for metal thread insert for both SM1 and side fixing hole

szymons90, 2023-05-07:
No, I was thinking about leaving the SM1 thread in the metal element (2). This base (1) could be printed with a standard M4 threaded insert installed.
Sorry, the M4 thread was missing from the project, in the second screen it should be visible. Of course, thickness in this region shall be slightly enlarged

[image]

[image]

gkasprow, 2023-05-07:
are you sure Thorlabs is using M4 thread? I thought it was some imperial standard

szymons90, 2023-05-07:
From what I see for mounting fixtures they use either M4 or 8-32 - below I send an example link:

https://www.thorlabs.com/drawings/6c7f99117867bfa3-AA0B5334-E89D-E8F4-0EC272DA9CFEEE88/LMR1_M-AutoCADPDF.pdf

gkasprow, 2023-05-07:
Ok, makes sense, I have US version :)

szymons90, 2023-05-08:
Ok, take a look at the V3. Nylon-printed Base (I had to thicken it to add the necessary material for threaded inserts) and aluminum Housing, made just by turning. As the thickness of the base increased, the height of the housing decreased, thus the entire thing should be a little bit cheaper, and lighter than V2. The Base flange is relatively thick, but I don't think that's a problem, as the nylon has less density than aluminum. But we have to pay attention to the M1,6 mounting screws for PCB - there are not such small inserts on the market, thus we will have to use self-tapping screws - but to be honest I don't think that it will be a problem. We can also check taps for plastics - as far as I know, there are such things on the market.

[DiPho_V3_20230508.zip](https://github.com/sinara-hw/DiPho/files/11424579/DiPho_V3_20230508.zip)
![DiPho_V3 1](https://user-images.githubusercontent.com/132475730/236917610-31f15f1a-dcbd-4805
[image]
-ba5e-22
[image]
88f4cf20d2.PNG)

szymons90, 2023-05-08:
Section view didn't load correctly, here it is:

[image]

szymons90, 2023-05-08:
Hmmm, I've got two ideas at the moment:
1. To use the side surface of the PCB - but I don't think that would work.
2. To replace two bushings with a single 3D printed ring with holes, which will be located between two PCBs and supported on a larger surface. Such solution will make these PCB as some kind of sandwich structure

szymons90, 2023-05-10:
Ok, take a look at the V4 version. It's all 3D printed. As we talked yesterday, we can glue the retaining ring for 1 inch tube to the housing. This would allow to attachment of a standrad 1 inch tube. The only thing that worries me relates to the length of such a ring - it's just 2 mm. If needed, we can use standard plastic rings, which have 2,8 mm in length.

[image]
[image]
 
[DiPho_V4_20230510.zip](https://github.com/sinara-hw/DiPho/files/11438531/DiPho_V4_20230510.zip)

szymons90, 2023-05-10:
Also, as I mentioned before, I replaced two brass spacers, by a single one ring, which allow to keep perpendicularity of the PCB surface. In this version, there is no custom metal parts.

gkasprow, 2023-05-10:
LGTM. I'd move the threaded ring by 0.5mm from the edge

szymons90, 2023-05-29:
Hi,

Just finished an integration of the first prototype. IMHO it looks OK, but there are still a few things I'd like to improve in the next versions:
1. In my opinion, the mounting holes of the M1.6 in PCBs are unnecessary. Both PCBs are well positioned on the cutouts on the connectors, and since there is a ring between the plates, they also hold reliably. There are no such small inserts, so to use these holes, we would have to either thread them or use some self-tapping screws. Additionally, there will be a little more space for components.
2. I have to increase the cut-outs in which boards are placed because I did not take into account that after cutting the PCB, small teeth remain on the edges. There is no point in grinding them, increasing the diameter by 1.5 / 2 mm should do the trick.
3. I also need to increase the clearance between the ring and the housing a bit, after printing it is quite tight.
4. The patent with the snap ring to the lid works well. We didn't even buy UV glue, we used plain cyanoacrylate for now.
5. I used an additional o-ring between the PCB with the diode and the cover - so that, in case of emergency, no light gets in from the side of the slots with connectors. We can get rid of it if necessary, but it costs 50 gr each. This also makes it a bit more secure. They are presented in the section view below.

Here are some photos. In my opinion, if it was painted black and the screws were blackened, it would look very nice. Let me know what do you think.
[image]
[image]
[image]
[image]
[image]

gkasprow, 2023-05-29:
yes, painting is a must.
Can we place text on the back side?
otherwise we would have to print and stick a label...

gkasprow, 2023-05-29:
otherwise it looks great

szymons90, 2023-05-30:
There should be no problem with a text - let me know what it should be and if there is any font or logo.

There is also an issue of LED pipes - we are thinking about getting rid of them. They are relatively expensive (PLN 5 / piece), which is an unnecessary cost regarding this size of casing. Besides, since it will be used indoors, it doesn't have to be 100% sealed from the connector side. And they are still problematic to assemble... We would just prefer to slightly modify the housing in this place, to make some kind of lip, so that it would almost stick to the diodes. What do You think?

gkasprow, 2023-05-30:
Text:
SPE close to a 2-pin connector
TERM near to the switch
OUT close to the coax connector
and USB

and, of course, DiPho v1.1

I would also add some plastic around USB and SPE connectors so they are more visually part of the enclosure
about LEDs - the hole would be fine, just add some lips or tubes around so the light is not scattered inside the enclosure.

szymons90, 2023-05-31:
Sth like that? Maybe I'll try to make the font sligthly bolder, to increase the text quality after printing.
[image]

szymons90, 2023-05-31:
Ok. What about the font and its size? Is it OK?

gkasprow, 2023-05-31:
yes

szymons90, 2023-05-31:
Better? The problem with the coax connector is that it is lower than the SPE and USB, so i made a groove around it.
Are the subtitles ok or would you like all of them horizontal?

[image]

gkasprow, 2023-05-31:
Do we need this groove around coax comnectot?

szymons90, 2023-05-31:
From my opinion, it would be the best to leave the space around this connector free. It barely sticks out of the casing.
[image]
[image]

szymons90, 2023-09-04:
Finally, the painter finished these housings (2 months... - no comments). IMHO it looks pretty good, although the color was supposed to be matte black, and it is slightly glossy.

Overall:
- IMHO there is no point in playing with convex letters, the prints look cool.
-  I would also get rid of flanges protecting the connectors, I would make the whole bottom flat, so that it would print comfortably.
- there's also nothing to play with LED pipes, but we've established that before.

Let me know what do You think.

[image]
[image]
[image]
[image]

gkasprow, 2023-09-04:
looks good to me

pm-daniel, 2023-09-04:
As we discussed - separate LED's should be merged into bicolor one, of bigger size in the next revision. That would simplify that part of the design.

During bring up I found few other electronic design issues, I will describe them soon in separate thread.

szymons90, 2023-09-04:
No problem, the technology is under control, so any changes to a serial product will not be problematic. 
As for the LED holes, yes, I do remember, a single one will definitely be better.
Just let me know when you have the stepper ready